fishinabarrel/linux-kernel-module-rust #1: File a bug with Rust/Cargo requesting the ability to ship a target.json in a crate

alex, 2018-05-15:


geofft, 2018-06-03:
It looks like the ability to ship / install arbitrary resources in crates is rust-lang/cargo#2729 and rust-lang/rfcs#2376 (both still open). The latter is specifically about installing things; the former seems to also encompass resources used by the application. You can already include arbitrary files in your crate (via `includes` in `Cargo.toml`) but they only end up in the crate source package, i.e., they're only really useful for build scripts or `include_str!` or similar.

I looked a bit at whether it's possible to implement this in cargo via a new build script option, like `cargo:rustc-target-json={"disable-redzone": true, ...}`. But then I realized that if that's doable, it should be possible to implement this _already_ by having the `linux_kernel_module` build script create the `target.json` file inside `$OUT_DIR`, and then print `cargo:rustc-env=RUST_TARGET_PATH=...`.

That doesn't work, because cargo wants to have the target config already in place when it starts up, so that it can find out the names of output files (it runs `rustc --print=file-names` and appends a `--target` option if you specify one), even before it runs the build scripts. I'm not totally sure why it needs this, but there is probably a good reason and changing it is likely to be hard.

So, I think if this is going to work, it's going to need to pull the target specification out of the sources. I think this is reasonably easy: there's a `cargo metadata` command, and you can just go through `dirname(i['manifest_path']) for i in metadata['packages']` and see if any of those have the target.json. `cargo metadata` doesn't even take a `--target` argument, so it definitely works.

This could happen either in `cargo-xbuild` or `cargo` itself ... it's probably an easy source-level change in `cargo`, but it's definitely doable from `cargo-xbuild` too.

alex, 2019-09-01:
This is being obviated by simply upstreaming the target files.

closed 2019-09-01 by alex