fishinabarrel/linux-kernel-module-rust #3: Decide what to do about `malloc()` in filesystem::register

alex, 2018-05-16:
Upside: it works cleanly and has a nice API

Downside: extra malloc, having the `file_system_type` in heap memory means it's writable which means the function ptrs in it can be targets for control flow hijacking.

Possible solutions are: using a macro to convert the trait implementer to a static, mapping the memory read-only, or deciding it's all fine since our rust modules won't be full of exploitable vulnerabilities (<--- probably not this).

geofft, 2018-05-16:
Can we make a const fn constructor for Filesystem that creates the file_system_type, and only impl register on &'static Filesystem?

alex, 2018-05-16:
Ooooh, love it.

alex, 2018-05-19:
Abandoning this:

- a) the memory was never mapped RO, because there's an intrusive linked list and a lock and some other crap in `file_system_type`, so we're not making the function-pointers-available-for-hijacking problem any worse
- b) There's way too much stuff we need that's not a `const fn`, so that approach isn't tractable ATM.

closed 2018-05-19 by alex