fishinabarrel/linux-kernel-module-rust #13: Add support for registering sysctls

alex, 2018-05-24:
Rough cut concept for the API:

```rust
struct MySysctlExample {
    sysctl_table: SysctlTableRegistration,
}

impl KernelModule for MySysctExample {
    impl init() -> KernelResult<Self> {
        let table = SysctlTable::new();
        let sub_table = SysctlTable::new();
        sub_table.register(INTEGER_VALUE, "name", some_parameter_about_valid_values_and_type?);
        table.register_subtable(INTEGER_VAULE, "name", sub_table);
        
        return Ok(MySysctlExample{
            sysctl_table: table.register();
        });
    }
}
```

Feedback welcome, going to spike on this tomorrow

geofft, 2018-05-25:
I have a longer API suggestion about accessing the data that I'll post once my other laptop charges, the tl;dr is a trait with `&self` methods for reading and writing data that you can implement on anything with interior mutability (`AtomicU32`, `RwLock<String>`, etc.), because multiple threads can try to read and write the sysctl at once.

But re registering - torvalds/linux@f728019b says that the `child` pointer in `struct ctl_table` is deprecated, and the way you're supposed to use the interface is not by making nested `struct ctl_table`s, it's by having a single one (which can include multiple sysctls) and calling `register_sysctl_paths`, which also takes a list of strings describing the path where you want to register them (in the form of a list of `struct ctl_path { char *procname; }`s, for some reason). As far as I can tell most code in the kernel hasn't gotten the message because that commit also provides backwards-compatibility for existing callers, but e.g. [the end of kernel/seccomp.c](https://github.com/torvalds/linux/blob/v4.16/kernel/seccomp.c#L1244-L1279) is a good example of the new API.

So, I think we do want the register function to take a sysctl path, either in the form of a single string that we split on slashes/dots or an `&[&u8]`, and avoid nesting `SysctlTable`s. I'm not sure if we want the ability to register multiple sysctls at once; I think it's fine to just loop through the sysctls, because Rust makes it hard to have a structure that owns elements of different concrete types, but much easier than C to loop through things, even things of different concrete types (`for i in &[&my_atomic_int as &Sysctl, &my_string, &my_other_type...]`).

geofft, 2018-05-26:
API suggestion for the read/write side of this:

```rust
trait TranslateSysctl<T> {
    fn get(&self, var: &T, w: &mut core::fmt::Write) -> KernelResult<()>;
    // requires shared/interior mutability on var, because multiple threads
    // can try to write to a sysctl at once.
    fn set(&self, var: &T, buf: &[u8]) -> KernelResult<()>;
}

struct U32MinMaxTranslator(Option<u32>, Option<u32>);
impl TranslateSysctl<AtomicU32> for U32MinMaxTranslator {...}
struct ValidPidTranslator;
impl TranslateSysctl<AtomicU32> for ValidPidTranslator {...}
...

struct GenericTranslator; // or we can just use ()
impl TranslateSysctl<AtomicBool> for GenericTranslator {...}
impl TranslateSysctl<AtomicU32> for GenericTranslator {...}
impl TranslateSysctl<RwLock<String>> for GenericTranslator {...}
...

struct Sysctl<T, U: TranslateSysctl<T>> {
    pub inner: T,
    translator: U,
    registration: *mut ctl_table_header,
}
impl<T, U> Sysctl {
    fn register(name: &[u8], var: T, translator: U, mode: mode_t) -> KernelResult<Self> {
        Sysctl { inner: var,
                 translator: translator,
                 registration: unsafe { register_sysctl_paths(...)? }, }
    }
}
impl Drop for Sysctl {
    fn drop(self) {
        unsafe { unregister_sysctl_table(self.registration) }
    }
}
```

which you'd use like this:

```rust
struct MySysctlExample {
    verbose_log: Sysctl<AtomicBool, GenericTranslator>,
    timeout: Sysctl<AtomicU32, U32MinMaxTranslator>,
}

impl KernelModule for MySysctlExample {
    fn init() -> KernelResult<Self> {
        let v = Sysctl::register(b"fs.rustfs.verbose_log\0",
                                 AtomicBool::new(false),
                                 GenericTranslator,
                                 0o644);
        let t = Sysctl::register(b"fs.rustfs.timeout_secs\0",
                                 AtomicU32::new(4),
                                 U32MinMaxTranslator(0, 60),
                                 0o644);
        MySysctlExample { verbose_log: v, timeout: t }
    }
}

impl MySysctlExample {
    fn do_something(&self) {
        if (self.verbose_log.inner.load(Ordering::Relaxed)) {
            printk!("rustfs: doing the thing");
        }
        do_the_thing();
    }
}
```

and you wouldn't need to write any more code as a consumer: the `TranslateSysctl` impls already exist (unless you need custom interpretation), and the `Drop` impls on both the kernel module and the sysctl take care of things the way you want. (I think - does this API actually work or have I messed up ownership somewhere?)

I wanted to do a `impl<T, U> DerefMut for Sysctl<T, U> { type target = T; }` but the docs now say you shouldn't use that except for smart pointers. So I guess we have to write `verbose_log.inner.load` and not `verbose_log`.

I tried a design where the registration just holds a borrow to the data item instead of owning it, but that gets into the problem that you can't store the registration and the thing it's borrowing in the same structure, which is a thing probably everyone is going to want to do. (Well, I guess there's [rental](https://github.com/jpernst/rental) but that seems excessive, and this design seems fine I think.)

One thing this doesn't address is permissions beyond what the mode bits give you. Linux handles permissions via a global variable for the current process, which is probably what we want to do too - have a Rust binding to `capable(CAP_SYS_ADMIN)` or whatever and allow custom impls of `TranslateSysctl` (and everything else) to just call that, instead of trying add some permissions structure as an argument to every function. (Also see torvalds/linux@bfdc0b49 for an interesting edge case re the mode bits.)

alex, 2018-05-26:
Overall this looks like a great improvement on what I suggested, big 👍 !

Important questions:

1) In your examples you pass the sysctl name as a `.` seperated string, rather than as an array of path components. What's the lifetime the kernel needs on the `ctl_path[]`? If it's ephemeral, it's fine to just split the string into a local array, if it's longer (i.e. same length as the registration) we also need to keep that live. If you don't know this offhand I can go cave diving in the kernel source.

2) The example you linked in seccomp registers both sysctls under the `kernel.seccomp` namespace in the same `register_sysctl_paths` call. Is it fine to register them in two separate calls? Or does the entire namespace need to go at once?

3) These don't seem to involve mibs at all. Are those not a thing anymore?

4) `inner` probably needs to be `Sync` yeah? Also should we be requiring the `KernelModule` type to be `Sync` (Maybe not, nothing uses it besides the destructor)

geofft, 2018-05-26:
1. I don't know off-hand. Also, I'm not sure a single string is the right plan, really; it's just that a series of null-terminated strings is annoying without a macro (there's https://github.com/mzabaluev/rust-c-str or https://github.com/rust-lang/rfcs/issues/400#issuecomment-274140470 or something). The second-most common pattern after a literal, I think, is the thing where the network subsystem creates one directory for each network device, so maybe we actually want to support `&[c!("net"), c!("ipv4"), c!("conf"), device.name, c!("arp_filter")]` for that... at which point you care about lifetimes. (My guess would be that the string gets copied into a dentry, but I'd have to check to be sure.)

2. I believe it is fine on the grounds that a) you can register multiple sysctls in _different_ paths and the kernel knows how to merge them b) the backwards-compat code I links above turns one `register_syscall_table` call into multiple ones. Easy to test, too.

3. They're not a thing on Linux. Linux does have a [`sysctl`(2)](http://man7.org/linux/man-pages/man2/sysctl.2.html) that takes a vector of constants, but "use of this system call has long been discouraged, and it is so unloved that it is likely to disappear in a future kernel version." The filesystem API showed up within a year of this one, I think. Also, the constants have to be global and unlike MIBs they're not OID arcs, so there's no reasonable way to use them from an out-of-tree module anyway, I think.

4. Oops, yes re `inner`. I think we don't for `KernelModule` since we don't pass a C pointer to it around (the way we pass a C pointer to `inner` to the sysctl code) - if someone has a pattern where they share it, and it autoderives `Sync`, great, but the Linux APIs don't require it. (If we end up making an actual Rust wrapper for `__this_module()`, we'll want to think about thread-safe accessor APIs for that.)

geofft, 2018-05-26:
Also:

- I originally had a design where you didn't have an explicit translator object and you just did `impl SysctlVar for AtomicU32` etc. That gets you more concise registration syntax, at the cost of having no place for validators. I think if we want a concise syntax we should just use `()` instead of `GenericTranslator`

- Probably we want something like this in general for all sorts of variables, like procfs, sysfs, configfs, module parameters (both on the `modprobe` command line and in `/sys/module/$module/parameters/`), etc.

alex, 2018-05-26:
I was planning on implementing it as `register` and
`register_with_validator`, and just using a translator in the first call,
to keep it less verbose in the common case :-)

commit cf3212cb (2018-06-10): Fixes #13 -- added initial sysctl API (#17)
closed 2018-06-10 by alex