fishinabarrel/linux-kernel-module-rust #30: Remove explicit list of symbols in build.rs?

alex, 2018-05-28:
It's a pain to have to constantly update it. Is there a reason not to just bind everything -- other than the compile time penalty (which should theoretically happen less often since we'll be editing `build.rs` less?)

alex, 2018-05-30:
This cannot be done until https://github.com/rust-lang/rust/issues/51192 is resolved, or const generics land.

alex, 2019-06-09:
https://github.com/rust-lang/rust/issues/61415 is now the blocker for this.

alex, 2020-08-09:
Attempted again in https://github.com/fishinabarrel/linux-kernel-module-rust/pull/234 -- most of the impls now exist for arrays longer than 32... except `Default`, which is still limited. (This is slightly messier than other impls, because `[T; 0]` currently impls `Default`, even when `T` does not!)

damageboy, 2020-08-25:
In the absence of this, would be correct to assume that if I want to add some kernel functionality for my driver the correct way is still to update `build.rs? Are you accepting PRs for it etc, or are we in some sort of limbo state?

Could there be, alternatively, an disallow-list for stuff that is knowingly wrong?

You last message on this, and attempt with #234 seems to imply this could be 90% OK-ish, with only small chunks missing, is that the case?

alex, 2020-08-25:
Yes, adding new bindings is done with a PR to `build.rs`.

We're definitely still accepting new PRs.

If bindgen has an API to do deny-list instead of allow-list, that'd be fine with me.