fishinabarrel/linux-kernel-module-rust #37: Make it possible to write modules in stable Rust

geofft, 2018-06-01:
Even though you'll need the unstable compiler for a while for this crate, modules themselves should be possible to write in stable Rust, in the same way that libstd itself requires unstable but can be used from stable Rust.

Right now `hello-world` uses `#![feature(alloc)]` to `use alloc::borrow::ToOwned` and `alloc::String`. Both `alloc::borrow` and `alloc::string` are stably re-exported in libstd. So we should do the same thing. Strictly speaking, this permits those modules to change as long as libstd makes an API-compatible facade, but practically that's unlikely to happen and if it does we can just steal whatever facade libstd comes up with (or in the worst case, take a semver hit).

Potentially the way to do this is to re-export a module named `std` that contains a subset of what's in actual libstd, with the intention of consumers doing

```
#[macro_use]
extern crate linux_kernel_module;
use linux_kernel_module::std;
use linux_kernel_module::std::prelude::v1::*;
```

That way, those things from std that do/can exist in kernelspace can just be used like normal.

alex, 2018-06-01:
I wonder if we should file a bug with cargo-xbuild asking for the ability
for a target to provide an alternate prelude. Would that make sense?

geofft, 2018-06-01:
I tried playing around with `#[feature(prelude_import)]` but I couldn't get it to work... probably it only works with libcore and libstd specifically, I didn't source-dive rustc hard enough to find out.

A slightly more compelling thing would be for a target to provide an alternate _libstd_, which means that third-party crates that aren't `no_std` because they need liballoc could work (collection types, things that use a `Vec`, etc.)

geofft, 2019-09-08:
Since `alloc` is stable, we're now down to just `#[feature(const_str_as_bytes)]` for modinfo. Keep in mind that modinfo doesn't even work in its present form anyway (#80)....

alex, 2019-09-08:
That feature is going stable soon fwiw

geofft, 2020-08-09:
Oh hey, this has been fixed since #207!

closed 2020-08-09 by geofft

geofft, 2020-08-13:
Well, I forgot that `-Z build-std=core,alloc` is unstable. But that feels very similar to the fact that you need nightly for the linux-kernel-module-rust dependency itself, I think it's enough to say that there's no feature gates in the source code of modules.