fishinabarrel/linux-kernel-module-rust #42: bindgen derives Debug on a packed struct

geofft, 2018-06-03:
```
ubuntu@ip-172-16-0-28:~/linux-kernel-module-rust/hello-world$ RUST_TARGET_PATH=$(pwd)/.. cargo xbuild --target x86_64-linux-kernel-module                                                                                                                                          Compiling linux-kernel-module v0.1.0 (file:///home/ubuntu/linux-kernel-module-rust)                                                  warning: #[derive] can't be used on a non-Copy #[repr(packed)] struct (error E0133)
    --> /home/ubuntu/linux-kernel-module-rust/hello-world/target/x86_64-linux-kernel-module/debug/build/linux-kernel-module-ba51b8057957ede2/out/bindings.rs:2455:10
     |
2455 | #[derive(Debug, Default)]
     |          ^^^^^
     |
     = note: #[warn(safe_packed_borrows)] on by default
     = warning: this was previously accepted by the compiler but is being phased out; it will become a hard error in a future release!
     = note: for more information, see issue #46043 <https://github.com/rust-lang/rust/issues/46043>
```

tl;dr: if packing a struct causes the field to get misaligned, accesses are unsafe, and the derived implementation of `Debug` involves accessing these fields.

The struct in question is `desc_struct` (which is the type of a member of `thread_struct`, which we need).

Do we care? Also I think this is a bindgen bug and we can't do anything about it?

alex, 2018-06-03:
Did you add something to the `build.rs` whitelist to provoke this?

geofft, 2018-06-03:
Nope - this was with current master (f1150b84079d3a2eb3ea42a004027a51784c72a2) on the 18.04 EC2 box.

```
ubuntu@ip-172-16-0-28:~/linux-kernel-module-rust/hello-world$ cargo --version
cargo 1.27.0-nightly (9e53ac6e6 2018-05-07)
ubuntu@ip-172-16-0-28:~/linux-kernel-module-rust/hello-world$ cargo xbuild --version
cargo-xbuild 0.4.2
ubuntu@ip-172-16-0-28:~/linux-kernel-module-rust/hello-world$ rustc --version
rustc 1.27.0-nightly (ff2ac35db 2018-05-12)
ubuntu@ip-172-16-0-28:~/linux-kernel-module-rust/hello-world$ grep bindgen Cargo.lock
name = "bindgen"
 "bindgen 0.36.1 (registry+https://github.com/rust-lang/crates.io-index)",
"checksum bindgen 0.36.1 (registry+https://github.com/rust-lang/crates.io-index)" = "acc8fa644dc2b6f6c2d391ccea460da1ad9bf2271c4e9ffaad88d2dd727a7295"
```

alex, 2018-06-03:
I guess it's an 18.04 kernel thing?

Can you try blacklisting that name?

alex, 2019-06-09:
We blacklisted this.

closed 2019-06-09 by alex