fishinabarrel/linux-kernel-module-rust #43: Test on i386

geofft, 2018-06-03:
While looking at something else I remembered that the i386 kernel builds with `-mregparm=3 -freg-struct-return`, i.e., pass the first three arguments in registers (instead of on the stack, the normal ABI) and try to return structures in a register if they fit. I'm not sure if Rust and/or bindgen know how to deal with this.

If they don't, put a note in README.md and move on. If they do, let's add i386 to CI at some point....

alex, 2018-06-03:
Before we test on i386 we need to fix c_types.rs because it is 100% x86_64
specific ATM.

alex, 2019-09-13:
Same checklist as #165:

- [ ] Upstream a target to Rust (including obviously figuring out all the details of that ABI)
- [ ] Figure out how we can test it in CI (chroot?)
- [ ] Add support to c_types.rs
- [ ] Turn it all on!