fishinabarrel/linux-kernel-module-rust #60: Come up with a shorter idiom for handling errors in C calls

alex, 2018-07-04:
Right now error handling for C functions is fairly verbose:

```rust
let res = unsafe { bindings::call() };
if res != 0 {
    return Err(error::Error::from_kernel_errno(res));
}
```

I think we want a `try!()`-style macro for checking the return value and propagating it upwards. The one challenge is not all kernel functions have the same return value convention, for some `!= 0` means error, for others `< 0` means error, and I'm sure there's some third convention.

geofft, 2019-08-16:
Was reminded of this while looking at #115. We can do this with a `#[repr(transparent)]` wrapper type on the various useful-value-or-errno types like ssize_t and pointers:

```rust
#![feature(try_trait)]

use std::convert::TryFrom;

#[repr(transparent)]
pub struct SSizeT(i64);

pub const E2BIG: SSizeT = SSizeT(-7);
pub const EINVAL: SSizeT = SSizeT(-22);

impl std::ops::Try for SSizeT {
    type Ok = i64;
    type Error = SSizeT;
    fn into_result(self) -> Result<i64, SSizeT> {
        if -4096 < self.0 && self.0 < 0 {
            Err(self)
        } else {
            Ok(self.0)
        }
    }
    fn from_error(v: SSizeT) -> SSizeT {
        v
    }
    fn from_ok(v: i64) -> SSizeT {
        if -4096 < v && v < 0 {
            panic!("value {} would be considered an errno", v)
        } else {
            SSizeT(v)
        }
    }
}

impl From<usize> for SSizeT {
    fn from(v: usize) -> Self {
        if let Ok(n) = i64::try_from(v) {
            SSizeT(n)
        } else {
            E2BIG
        }
    }
}

impl From<std::string::FromUtf8Error> for SSizeT {
    fn from(_: std::string::FromUtf8Error) -> Self {
        EINVAL
    }
}

extern fn foo() -> SSizeT {
    String::from_utf8(vec![65, 66, 67])?.len().into()
}
```

(see [playground](https://play.rust-lang.org/?version=nightly&mode=debug&edition=2018&gist=37c47b850a697fe3d19312b5055bac75), and try changing the bytes to invalid UTF-8).

Not sure if I like it. I think I do but I'm a bit unsure those are the right associated types and I'm also not sure I believe the `from_error` and `from_ok` impls. (Mostly I'm not sure where they're used ... replacing them with `panic!` works ....)

Note that 
* the inner field on `SSizeT` is private, so you can't accidentally construct something that will be misunderstood
* you can impl `From<some error type>` to map that to an errno in one place, which is nice.

alex, 2019-08-16:
This looks like it gives you `?` support for returning an `ssize_t` given a `Result`, I think this issue is about the opposite -- returning a `Result` given an `ssize_t` (or some other integer)

geofft, 2019-08-17:
That direction should work too:

```rust
use std::io;

impl From<SSizeT> for io::Error {
    fn from(v: SSizeT) -> io::Error {
        io::Error::from_raw_os_error(-v.0 as i32)
    }
}

fn bar() -> io::Result<i64> {
    Ok(SSizeT(-22)?)
}
```

[playground](https://play.rust-lang.org/?version=nightly&mode=debug&edition=2018&gist=ae7784ea648257547f4c38ae71bb611b)

Basically we need to use the `SSizeT` etc. wrapper types on the return values of imported C functions and exported Rust callbacks. Because it's transparent we can do that in the FFI definition itself and you don't need to actually say `SSizeT` in a function that uses it, just `Ok(some_kernel_function(...)?)`.