fishinabarrel/linux-kernel-module-rust #62: usercopy uses u64 on 4.15.0-1007-aws

geofft, 2018-07-12:
On the Ubuntu 18.04 EC2 box (which I still haven't shut down oops):

```
error[E0308]: mismatched types
   --> /home/ubuntu/t2/src/user_ptr.rs:103:17
    |
103 |                 data.len() as u32,   
    |                 ^^^^^^^^^^^^^^^^^ expected u64, found u32
help: you can cast an `u32` to `u64`, which will zero-extend the source value
    |
103 |                 (data.len() as u32).into(),
    |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0308]: mismatched types
   --> /home/ubuntu/t2/src/user_ptr.rs:129:17
    |
129 |                 data.len() as u32,   
    |                 ^^^^^^^^^^^^^^^^^ expected u64, found u32
help: you can cast an `u32` to `u64`, which will zero-extend the source value
    |
129 |                 (data.len() as u32).into(),                                                                                           |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                        
error: aborting due to 2 previous errors                                                                                                                                                                                                                                        ```

Changing it to u64 fixes it.

Probably something switched to `size_t` or whatever?

alex, 2018-07-12:
Hmmm, is the issue here that it's a different kernel, where the API takes a different type? Or that your box's arch is somehow different from where we test (I doubt it?)

geofft, 2018-07-12:
The box is regular `x86_64`. (ssh ubuntu@34.207.113.141, your key is on it) I'm guessing the API changed, but I haven't yet tracked down when, I just wanted to file this before I forgot.

alex, 2018-07-12:
Once we figure out how to make it pass,
https://github.com/alex/linux-kernel-module-rust/pull/63 should ensure we
don't regress again.

alex, 2019-05-05:
This got fixed recently.

closed 2019-05-05 by alex