fishinabarrel/linux-kernel-module-rust #64: usercopy handling and KERNEL_DS

thejh, 2018-08-28:
This project currently seems to assume that a pointer checked with `access_ok()` always points to userspace; however, thanks to the `set_fs(KERNEL_DS)` mechanism, that's not necessarily true.

`access_ok()` doesn't actually always check that the given pointer points to userspace. If you want real kernel memory safety around `access_ok()`, you'll have to split userspace access into two types:

- One that passes userspace pointers wrapped in opaque structs that can't be directly accessed from safe code, with helpers for safe slicing and such, just like for kernel buffers. This variant would also need to have object lifetime checking such that a handle to a "userspace" memory region can't be persisted beyond the end of the current function. (I don't know much about rust, but this seems like something rust probably provides?)
- One that accepts raw userspace pointers, but does a stronger check than `access_ok()` (comparison to `TASK_SIZE_MAX` or so). (Of course this would still permit corrupting userspace memory, but I assume that's acceptable.)

Longer explanation of the problem:

Linux has various pieces of kernel code that normally operate on userspace pointers, and are optimized for this case - for example, filesystem read and write handlers (`->f_op->read()`, `->f_op->write()`, ...). These handlers take userspace pointers (annotated with `__user` and, somewhat redundantly, accessed through helpers that check for `access_ok()`). However, sometimes the kernel needs to directly call into such APIs, operating on kernel buffers. In current kernel versions, this is in particular true for the `sys_splice()` syscall, which can move data directly between a pipe and a file without going through userspace memory. (There have been discussions about cleaning this up to make it less hazardous, but that hasn't happened yet.)

Normally, the `access_ok()` checks performed by filesystem code would prevent this from working - after all, these checks are explicitly designed to block the use of kernel pointers. To work around this, the kernel has a function `set_fs()` that can be used to temporarily override the address limit used for `access_ok()` checks so that all pointers are accepted. Therefore, if your code can run in a context where address limit checks have been disabled, `UserSlicePtr` could AFAICS be used to access arbitrary kernel pointers.

Kernel code that does this looks like this (example from `fs/read_write.c`):
```
ssize_t kernel_read(struct file *file, void *buf, size_t count, loff_t *pos)
{
	mm_segment_t old_fs;
	ssize_t result;

	old_fs = get_fs();
	set_fs(get_ds());
	/* The cast to a user pointer is valid due to the set_fs() */
	result = vfs_read(file, (void __user *)buf, count, pos);
	set_fs(old_fs);
	return result;
}
```

There were two recent LWN articles related to this issue:
 - https://lwn.net/Articles/760345/
 - https://lwn.net/Articles/762510/

Examples of kernel bugs caused by this behavior:
 - ARM/ARM64 arbitrary memory read via `access_ok()` in a perf subsystem interrupt, which could run in KERNEL_DS context: https://bugs.chromium.org/p/project-zero/issues/detail?id=822
 - Infiniband subsystem arbitrary kernel write via `copy_to_user()` in `f_op->write()` context: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=e6bd18f57aad1a2d1ef40e646d03ed0f2515c9e3
 - https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=31e33a5b41bb158f27c30e13b12d6e5e6513ea05
 - https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=60e6627f12a78203a093ca05b7bca15627747d81
 - https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=15279df6f26cf2013d713904b4a0c957ae8abb96
 - https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=a0341fc1981a950c1e902ab901e98f60e0e243f3
 - https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=f1e255d60ae66a9f672ff9a207ee6cd8e33d2679
 - https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=26b5b874aff5659a7e26e5b1997e3df2c41fa7fd

alex, 2018-08-28:
So to make sure I understand:

- `set_fs(KERNEL_DS)` effectively disables `access_ok` and other checks on pointers belonging to userspace. The intent of this is let you reuse some function, intended for userspace buffers, with a kernel buffer.
- If you had some code that ran in one of these contexts, and it read from a `UserspacePtr` _besides_ the trusted one, none of the well-formedness checks would occur, and that `Ptr` could be used to access arbitrary kernel memory.

So it is not the case that _all_ `UserspacePtr` are abusable (as I initially feared), but rather that the abstraction isn't as safe as we claim (booo!)

The simple example of some code in Rust that'd be vulnerable is if you implemented a `sysctl::SysctlStorage`, which used the data for the sysctl to find a ptr-value, and then constructed a `UserspacePtr` from that and read of it. In normal execution, this would be safe. However, when invoked by splicing `/proc/sysctl/path` the `SysctlStorage::store_value` runs in a `set_fs(KERNEL_DS)` context, and thus the `UserspacePtr` checks become no-ops, leaving you vulnerable to arbitrary kernel memory manipulation.

Linux makes me super angry.

thejh, 2018-08-28:
> set_fs(KERNEL_DS) effectively disables access_ok and other checks on pointers belonging to userspace. The intent of this is let you reuse some function, intended for userspace buffers, with a kernel buffer.

Yes, exactly.

> If you had some code that ran in one of these contexts, and it read from a UserspacePtr besides the trusted one, none of the well-formedness checks would occur, and that Ptr could be used to access arbitrary kernel memory.

Yeah.

> The simple example of some code in Rust that'd be vulnerable is if you implemented a sysctl::SysctlStorage, which used the data for the sysctl to find a ptr-value, and then constructed a [...]

Yeah. The API contract for filesystem write handlers is basically "the write handler must not access the calling process in any way, except for reading from the supplied userspace memory range". **But this is not just because of the splice stuff!**

Another thing that can go wrong is that if a file implementation violates this rule, an attacker could open a file descriptor to such a file and pass it to something like a setuid process (as stdout/stderr) or send it to some daemon over a unix domain socket. Then, the privileged process would write some data into the file descriptor, expecting the data to be treated as "just plain bytes"; if the file implementation then e.g. interprets the written data as a userspace pointer and writes to it, you potentially get kernel-facilitated in-userspace privilege escalation.

For this reason, if you think your read/write handler should be accessing the calling process in any way other than accessing the provided memory range, you'll have to either do things based on who called `open()` (since that signals the intent to use ambient authority to some degree, unlike `write()`), or you can have `ioctl()` handlers (which signal very clear intent).

This is also why some of the bandaid fixes for the linked issues were slighly more complicated. In particular, Infiniband checks for both `KERNEL_DS` (against kernel memory corruption) and for a change of credentials between `open()` and `write()` (to mitigate attacks based on passing affected file descriptors to unsuspecting higher-privileged processes). (But that's an ugly bandaid, not how things should work.)

alex, 2018-08-28:
🤯

geofft, 2018-08-28:
I think the bugs fall in three categories:

* Code that can be called in unexpectedly `KERNEL_DS` context (i.e., the caller was not intending to transfer control, and would have taken precautions if it kneew), like the perf interrupt handler. It looks like ARM fixed this by [restoring fs during the interrupt](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=e6978e4bf181fb3b5f8cb6f71b4fe30fbf1b655c)? Is this a thing we should harden, or is the bug solely with the ARM interrupt handler? Are there other cases like this besides perf interrupt handlers?
* Write (and read?) handlers that do excessively clever things with pointers, leading to a `KERNEL_DS` problem with `splice`, and a regular, non-Linux-specific problem if you exec a setuid program that writes messages that happen to be parseable by the excessively clever code. I'm sort of inclined to say "Don't do that." I wouldn't be opposed to a typesystem-level solution to block read/write handlers from doing silly things, but I don't think Rust gives us enough control to say "Closures of this type do not construct objects of that type" (but maybe it does, because `UnwindSafe` does this sort of thing?). I'm also a fan of solving this by making it sufficiently ergonomic to implement real ioctls in a well-specified way that people aren't tempted to be overly clever. (See https://github.com/alex/linux-kernel-module-rust/pull/12#issuecomment-392278352 for some thoughts here when we were trying to figure out if `UserPtr<T>` should be an affine type.)
* Write (and read?) handlers that overflow the buffer they were given. We weren't treating this as inherently unsafe, on the grounds that kernel code can validly mess with mapped userspace addresses whether or not userspace explicitly asks it to do so. But we should at least ensure that read/write handlers get real Rust slices, so the obvious thing to do is not to overflow the bounds (and intentionally overflowing the bounds requires doing excessively clever things with pointers).

Between the last two points I can also see an argument that constructing a `UserPtr<T>` should be `unsafe`. Compared to checking `get_fs() == USER_DS`, this has the advantage that, if you really know what you're doing, it _is_ valid to say `unsafe { set_fs(KERNEL_DS); let p = UserPtr::new(&something_in_kernel_memory); do_something(&p); set_fs(USER_DS); }`. Using a constructed `UserPtr<T>` would be safe.

That would force us to make sure none of our safe APIs involve expecting people to construct `UserPtr`s on their own, which is probably what we actually want. It does look like we currently only call `UserSlicePtr::new` once, and it's in an unsafe C callback anyway, so changing that to be unsafe would be easy.

amluto, 2019-05-04:
I think that a serious Rust kernel module API should try to avoid all this nasty legacy crud. If you want to write awful code that calls set_fs(), I think you should do it in C.

As a practical matter, I think this means that the kernel should add access_ok(), copy_from_user(), etc variants that unconditionally use USER_DS, and Rust should use that. And if you call into a Rust driver under KERNEL_DS, you lose.

alex, 2019-05-04:
I think my conclusion here, which I never explicitly wrote down, was that the correct solution was that `UserSpacePtr::new` should be unsafe, since you had to be constructing it with a pointer that came from a specific place, due to the `set_fs()` issue. Since the expectation was always that our bridge code would construct them since all the abstractions are implemented in this module, this shouldn't negatively impact users, while forcing us to be cognizant of the origin of pointers.

(It should also be noted that we don't yet work with the newer kernel APIs that got rid of the READ/WRITE flags)

amluto, 2019-05-04:
Hmm. That’s not such a bad idea.

There’s the downside that, if you call back and forth between C and Rust and get your fs changed, then the meaning of the pointer changes.  Same if there’s a context switch.

If an extra fat struct is considered acceptable overhead, UserspacePtr could carry an mm_struct pointer for added checking.

geofft, 2019-05-06:
We attempt to express "only valid within the current context" (i.e., you cannot yield or return from the syscall) in the typesystem using the Rust `Sync` trait—see the discussion in #54. Does it solve things enough if we require that `UserSpacePtr: Sync` (and also that constructing one is `unsafe`)?

PR #95 (2019-06-11): Make UserSlicePtr::new unsafe
Fixes #64
closed 2019-06-11 by alex