fishinabarrel/linux-kernel-module-rust #83: Run hello-world through tests

geofft, 2019-05-06:
It uses `String` which none of the others do, and it has its own makefile. We should have coverage for it since it's the user-facing example.

alex, 2019-05-06:
Or maybe we should delete it?

geofft, 2019-05-06:
I think we should definitely have a copy-and-paste example, including a working kbuild-compatible Makefile. I'm fine with moving it into the tests directory and updating the README, but we should keep the Makefile.

Also hm, example-sysctl doesn't go through CI either, does it? That one seems pretty redundant with the sysctl test.

alex, 2019-05-06:
Ok, so let's get hello-world building, add it to tests, and remove
example-sysctl

geofft, 2019-05-06:
Possible approach: let hello-world be the template. If there's a file (mostly Makefile and Cargo.toml) that's in hello-world but not in some other test, it gets a copy of that file with `s/hello-world/$test/`.

That way hello-world gets to remain a copy-and-paste example but we also don't get duplication of Makefiles etc.

alex, 2019-05-06:
I think that's more complexity than is needed -- tests already have a
slightly different Makefile.

I think the idea outcome would be to make the Makefile parameterized enough
that there was only one of them ever.