fishinabarrel/linux-kernel-module-rust #110: Support things that would be required for binderfs

alex, 2019-07-15:
- [ ] Allocate chrdev region with specific number of minors but not register devices (https://github.com/torvalds/linux/blob/v5.0/drivers/android/binderfs.c#L558-L559)
- [ ] Expand filesystem support (https://github.com/torvalds/linux/blob/v5.0/drivers/android/binderfs.c#L546-L551)
- [ ] Support `inode_operations` (https://github.com/torvalds/linux/blob/v5.0/drivers/android/binderfs.c#L459-L463)
- [ ] Support `super_operations` (https://github.com/torvalds/linux/blob/v5.0/drivers/android/binderfs.c#L341-L346)
- [ ] Support `nonseekable_open` on `FileOperations` (https://github.com/torvalds/linux/blob/v5.0/drivers/android/binderfs.c#L375)
- [ ] Support `ioctl` on `FileOperations` (https://github.com/torvalds/linux/blob/v5.0/drivers/android/binderfs.c#L376-L377)
- [ ] Support `llseek` as a noop on `FileOperations` (https://github.com/torvalds/linux/blob/v5.0/drivers/android/binderfs.c#L378)

geofft, 2019-08-24:
For the record, I can't type, the reference to #110 in 521e30c should be #118.

brauner, 2019-09-13:
Ok, I'm not sick anymore so I'm going to be around finally to help out. :) Sorry about that!