fishinabarrel/linux-kernel-module-rust #112: Architecture support (tracking issue)

geofft, 2019-07-24:
Rust supported architectures: https://forge.rust-lang.org/platform-support.html For our purposes, we don't care much about Tier 1 vs. Tier 2, and even Tier 3 support is probably fine.

Another useful reference is Debian's arch status as of February 2018: https://alioth-lists.debian.net/pipermail/pkg-rust-maintainers/2018-February/001215.html (doesn't include all the arches the kernel supports)

arch | description | LLVM status | Rust status
-- | -- | -- | --
alpha | [DEC Alpha](https://en.wikipedia.org/wiki/DEC_Alpha) | [removed from LLVM in 2011](http://lists.llvm.org/pipermail/llvm-dev/2011-October/044594.html) |
arc | [Synopsys ARC](https://en.wikipedia.org/wiki/ARC_(processor)) | present | needs port
arm | [ARM 32-bit](https://en.wikipedia.org/wiki/ARM_architecture) | present | [ARMv5 and up is Tier 2](https://github.com/rust-embedded/wg/issues/25)
arm64 | ARM 64-bit aka AARCH64 | present | Tier 2
c6x | [TI C6000-series VLIW DSPs](https://en.wikipedia.org/wiki/Texas_Instruments_TMS320) | no LLVM port |
csky | [C-Sky](https://c-sky.github.io/) | no LLVM port |
h8300 | [Hitachi H8](https://en.wikipedia.org/wiki/H8_Family) | no LLVM port
hexagon | [Qualcomm Hexagon DSP](https://en.wikipedia.org/wiki/Qualcomm_Hexagon), part of Snapdragon SoCs | present | needs port
ia64 | [Intel Itanium](https://en.wikipedia.org/wiki/Itanium) | [removed from LLVM in 2009](http://lists.llvm.org/pipermail/llvm-dev/2009-July/024066.html)
m68k | [Motorola 680x0](https://en.wikipedia.org/wiki/Motorola_68000_series) | [port exists](https://github.com/M680x0/M680x0-llvm/), [proposed for merge](https://lists.llvm.org/pipermail/llvm-dev/2018-August/125317.html) | [work in progress](https://lists.debian.org/debian-68k/2019/02/msg00003.html)
microblaze | [Xilinx MicroBlaze](https://en.wikipedia.org/wiki/MicroBlaze) for FPGAs | [removed from LLVM](http://lists.llvm.org/pipermail/llvm-dev/2013-July/064126.html) |
mips | [MIPS](https://en.wikipedia.org/wiki/MIPS_architecture) (R3000 / MIPS I and up) | [present for R4000 / MIPS II and up](https://github.com/llvm/llvm-project/blob/master/llvm/lib/Target/Mips/MipsSubtarget.cpp#L94) | Tier 2
nds32 | [Andes AndeStar 32-bit](https://lwn.net/Articles/738426/) | [proposed for merge in 2017](https://lists.llvm.org/pipermail/llvm-dev/2017-January/108964.html) | needs port
nios2 | [Altera Nios II](https://en.wikipedia.org/wiki/Nios_II) for FPGAs | [removed from LLVM in January 2019](http://lists.llvm.org/pipermail/llvm-dev/2019-January/129121.html) |
openrisc | [OpenRISC](https://en.wikipedia.org/wiki/OpenRISC) | [out-of-tree backend](https://en.wikipedia.org/wiki/OpenRISC) | needs port
parisc | [Hewlett-Packard PA-RISC](https://en.wikipedia.org/wiki/PA-RISC) aka hppa | no LLVM port |
powerpc | [PowerPC / POWER](https://en.wikipedia.org/wiki/PowerPC) 32-bit, 64-bit LE, 64-bit BE | all present | all Tier 2
riscv | [RISC-V](https://en.wikipedia.org/wiki/RISC-V) | present | Tier 2
s390 | [z/Architecture 64-bit](https://en.wikipedia.org/wiki/Z/Architecture) aka s390x | present | Tier 2
sh | [Hitachi SuperH](https://en.wikipedia.org/wiki/SuperH) / J2 | [experimental out-of-tree backend](http://lists.j-core.org/pipermail/j-core/2017-February/000509.html) | needs port
sparc | [Sun SPARC](https://en.wikipedia.org/wiki/SPARC) 32- and 64-bit | present | Tier 2 for 64-bit (SPARC v9), needs 32-bit port
um | [User-Mode Linux](http://user-mode-linux.sourceforge.net/) (on any userspace arch) | N/A | might just work?
unicore32 | [Peking University Unicore](https://en.wikipedia.org/wiki/Unicore) | no LLVM port |
x86 | [Intel x86](https://en.wikipedia.org/wiki/X86) aka i386 and x86-64 aka amd64 | present | Tier 1
xtensa | [Tensilica Xtensa](https://en.wikipedia.org/wiki/Tensilica), used on ESP microcontrollers | [proposed for merge in March 2019](http://lists.llvm.org/pipermail/llvm-dev/2019-March/130796.html) | [work in progress](https://github.com/MabezDev/rust-xtensa)

Re ARMv4 - the linked issue implies ARMv4 is only a microcontroller for OS-less use, but arch/arm/mach-gemini/ looks like an actual Linux target for ARMv4. There are also a few ARMv4T boards (i.e., with Thumb support). It looks like [LLVM wants to emit BLX instructions](https://lists.llvm.org/pipermail/llvm-dev/2019-March/131197.html) which switch to Thumb mode, so unclear if plain ARMv4 will work. (Though forcing it to emit non-Thumb code only doesn't sound like it should be too difficult....)

Re MIPS I - see simias/psx-sdk-rs#1. MIPS I has load delay slots, LLVM's codegen would need to be taught about it.

Re unicore32: https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg1617144.html argues in favor of removing it (apparently there is no public GCC source either??), but that doesn't seem to have been followed up on.

Some possible approaches for merging into mainline and dealing with architectures with no LLVM support and no realistic plans (e.g., processors no longer manufactured):
* Convince Linux to drop them <_< >_>
* Only support certain architectures via Kconfig - it's probably okay to only support some modules on the major architectures
* Use mrustc to compile via C https://github.com/thepowersgang/mrustc
* Use the LLVM C backend to compile via C #111

XVilka, 2019-08-21:
Note, Hexagon was added in the Rust compiler: https://github.com/rust-lang/rust/pull/62814
So might me just worth testing the support.

androm3da, 2019-10-08:
> Note, Hexagon was added in the Rust compiler: [rust-lang/rust#62814](https://github.com/rust-lang/rust/pull/62814)
> So might me just worth testing the support.

My work to get `hexagon-linux` support for rust is incomplete, I still have to work out changes to `libc` and possibly more beyond that.

But the good news is that codegen support for the hexagon processor in general has been present in the compiler for a while.  My recent changes were only for making userspace binaries. AFAIK the codegen support that was added by @michaelwu (around https://github.com/rust-lang/rust/pull/41524) should suffice for kernel module development.

geofft, 2020-08-09:
Unicore has been removed in torvalds/linux@05119217a9bd199c4b8b12c01f86df09108b109b.

geofft, 2020-08-27:
@nickdesaulniers says arc, hexagon, riscv, s390, and sparc are currently broken in ClangBuiltLinux CI (although support exists). I wonder if the breakage would affect Rust.

sanxiyn, 2020-08-30:
C-SKY LLVM port got proposed in [August 2020](http://lists.llvm.org/pipermail/llvm-dev/2020-August/144481.html).

nindalf, 2021-01-12:
> Convince Linux to drop them <_< >_>

Removing support for some of these architectures from the Linux Kernel is currently being discussed - [Old platforms: bring out your dead](https://lore.kernel.org/lkml/CAK8P3a2VW8T+yYUG1pn1yR-5eU4jJXe1+M_ot6DAvfr2KyXCzQ@mail.gmail.com/)