fishinabarrel/linux-kernel-module-rust #123: Build fails on v5.0+ against kernels built with gcc

geofft, 2019-08-18:
Since torvalds/linux@e9666d10a5677a494260d60d1fa0b73cc7646eb3, detection for `asm goto` (which gcc supports and clang does not) got moved from a check at each build to a check in the kernel config, so a kernel built with gcc will have CONFIG_CC_HAS_ASM_GOTO, which breaks the build with clang / breaks bindgen's ability to parse the code with clang. (I haven't tested it, but I expect a kernel built with clang to work fine.)

See discussion in lizhuohua/linux-kernel-module-rust#1.

geofft, 2019-08-18:
Good news and bad news:
* Extremely recent versions of clang (the clang 9 snapshot on https://apt.llvm.org works) support asm gotos, so they do work fine.
* If you have such a version of clang, it breaks _older_ kernels because now the build time check thinks that clang supports asm goto but some of the ifdefs are wrong, or something? With kernel 4.19 I get "implicit declaration of function 'asm_volatile_goto'" in arch/x86/include/asm/jump_lable.h. Switching back to clang 7 from Debian stable works....

alex, 2020-08-09:
@geofft I think this got fixed somewhere along the way, I'm definitely building on 5.x kernels.

closed 2020-08-09 by alex

geofft, 2020-08-09:
Oh hey thanks for the reminder, I definitely regressed this last night while trying to get rid of kernel-cflags-finder and this probably explains why.

geofft, 2020-08-09:
@alex so, there is one particular case still broken - clang 9+ and kernel <5 (my last bullet point above).
https://travis-ci.com/github/fishinabarrel/linux-kernel-module-rust/builds/179047119
clang 9+ and kernel >=5 is fine. (clang <9 and kernel >=5 definitely won't work because of lack of asm goto support.)

Do we want to open a ticket for the new-clang-old-kernel case and see if it's solvable, or do we just want to document it as a restriction?

alex, 2020-08-09:
Just consider it a restriction I think.