fishinabarrel/linux-kernel-module-rust #134: Replace most (if not all) of the author fields with FiaB

alex, 2019-08-20:
This was it accurately reflects all the contributors we have/will have.

geofft, 2019-08-20:
We do have three other contributors: `git log -p --perl-regexp --author '^(?!Geoff|Alex)'`

Also, speaking of authorship, we should probably start enforcing Signed-off-by and link it to the upstream DCO so that we're in a good state for (potentially) upstreaming. I think it is fine for the two of us to attest to our contributions at some future point, but we should ask for it for any substantial outside work so we don't have to track people down later. (The existing contributions by not-us are textually small so I think we're fine.)

alex, 2019-08-20:
Is there some sort of git hook that will correctly enforce this? Because I do not want to do it manually.

alex, 2019-08-20:
And by git hook obviously I mean github webhook.

geofft, 2019-08-20:
`git rev-list "$TRAVIS_COMMIT_RANGE" | while read "$i"; do if ! git show --format=%ae -s | egrep -q '^(geofft@ldpreload.com|alex.gaynor@gmail.com)$' || git show -q --format=%B -s | grep 'Signed-off-by: '; then echo "Commit $i is missing a signoff" >&2; exit 1; fi` or something?

alex, 2019-08-20:
Heh, sure, maybe that works. I figured there was something prettier, but
that seems like it works well enough.

alex, 2020-08-09:
Ok, we still have a bunch of cargo author fields that aren't correct. The reason for this is that cargo author fields want an email address. Having an email address to go with "FiaB Contributors" is the blocker here.