fishinabarrel/linux-kernel-module-rust #139: Upstream support for building rust modules to mainline kernel

alex, 2019-08-20:
Kees thinks this is a good next step -- what we can do is upstream support for building rust with an arbitrary target file + xbuild, but not include any of our code (including how the build works).

We think we want to support something that looks roughly like:

```make
obj-m += helloworld.o
helloworld-crates := Cargo.toml

all:
        $(MAKE) -C $(KDIR) M=$(CURDIR)

clean:
        $(MAKE) -C $(KDIR) M=$(CURDIR) clean
```

alex, 2019-08-21:
New concept, inspired by the kernel's ASN.1 support (yeah, that's what you want to hear...)

```make
obj-m += helloworld.rust.o

all:
        $(MAKE) -C $(KDIR) M=$(CURDIR)

clean:
        $(MAKE) -C $(KDIR) M=$(CURDIR) clean
```

Things that are `.rust.o` would be assumed that `M` was a directory with a Cargo project. You should probably be able to optionally override that, but it'd be the default.

alex, 2019-08-22:
Started hacking on this here: https://github.com/alex/linux/pull/3/files

kees, 2019-09-06:
Why not include the linux-kernel-module code as part of it? That'd be needed by anyone doing kernel modules in rust. (Though it likely needs to be a separate module somehow... I haven't looked at how Rust handles shared objects...)

alex, 2019-09-06:
My thought was to keep the size of the individual pieces to land as small as possible: Landing build system support is smaller than landing the full SDK, so we should do build system only first.