fishinabarrel/linux-kernel-module-rust #151: Automated conversion from C to Rust

XVilka, 2019-08-30:
Idea is to provide the framework for automated conversion using [c2rust](https://github.com/immunant/c2rust), as they do for the user-space programs.

Corresponding issue in c2rust itself: https://github.com/immunant/c2rust/issues/150

geofft, 2019-08-30:
Interesting! I think the two things to do are
* use the `c_types` module in this crate instead of the `libc` crate (analogous to bindgen `builder.ctypes_prefix("c_types")`)
* make sure FFI calls go through `bindings`

Do you have an idea of what a good first target would be?

geofft, 2019-08-30:
Hmmmm, seems like c2rust handles macros just fine. I wonder if we want to use it to process src/helpers.c instead of compiling that as a separate C object. Since helpers.c is a separate translation unit, calls to macros can't get inlined and get a bit slow.

alex, 2019-08-31:
FWIW, calls across translation units are inlinable if we use LTO.

rinon, 2019-09-03:
Just wanted to pop over here to say we're definitely interested in making it easier to get c2rust working on kernel modules. We've translated one module so far, and things seems to work fairly well after we improved support for a bunch of gcc-isms.

Automatically refactoring the initial output of c2rust to use your types and bindings ought to be fairly straightforward and might be a good path to better usability.

XVilka, 2019-12-24:
Sorry for the noise, but were there any updates regarding this goal/blog post?

rinon, 2019-12-27:
We're still planning to put up a blog post on using C2Rust for a kernel module, hopefully in January.

marcograss, 2020-04-26:
is there any update on the blogpost for c2rusting a kernel module? I'm really looking forward to it

rinon, 2020-04-30:
Not yet, but thanks for the reminder. It had to get pushed back a bit because of other things taking priority. However, definitely still going to happen, hopefully can write it soon!

stappersg, 2020-10-25:
Which kernel module has been _c 2 rust_? Is it `virtio-net`?   ( Context #260 )

elichai, 2020-10-28:
FWIW last time I checked `c2rust` violates rust's aliasing rules (or proposed aliasing rules) although it can be solved by using raw references(https://github.com/rust-lang/rust/issues/73394) 

Stacked borrows(rust's proposed aliasing rules): https://github.com/rust-lang/unsafe-code-guidelines/blob/master/wip/stacked-borrows.md