fishinabarrel/linux-kernel-module-rust #163: Add API equivilant to kernel's nonseekable_open

alex, 2019-09-01:
Needed for #110

nelhage, 2020-08-28:
Does this mean:

(1) Somehow bind `nonseekable_open` so that you can call into it from Rust in your `open` implementation, or
(2) Somehow implement support for directly binding any of the common C `open` functions: https://github.com/torvalds/linux/blob/v5.8/fs/open.c#L1326-L1372

You could imagine (2) working by removing `open` from the `FileOperations` trait, and adding an `Open` trait, and in addition to the `open`  builder method, adding e.g. `open_nonseekable()` that binds to the kernel method.

alex, 2020-08-28:
Either! Would need someone to look into it.

alex, 2020-08-30:
The API for this is much more obvious with our new `FileOperations` API:

```rust
impl FileOperations for Foo {
    const SEEK: SeekFn<Self> = Some(open_nonseekable::<Self>);
}
```

alex, 2020-08-30:
Err, I apparently don't have brain cells to rub together. `open` is a function on the trait, not a function-ptr in a const on it. Which means this just needs to be a function that `open` delegates to. Functions calling functions is good.