fishinabarrel/linux-kernel-module-rust #167: build error: setfl binding missing

aurabindo, 2019-09-02:
```
error[E0560]: struct `bindings::bindings::file_operations` has no field named `setfl`
   --> /run/media/aj/Buffer/git-clones/linux-kernel-module-rust/src/chrdev.rs:228:13
    |
228 |             setfl: None,
    |             ^^^^^ `bindings::bindings::file_operations` does not have this field
    |
    = note: available fields are: `owner`, `llseek`, `read`, `write`, `read_iter` ... and 27 others

error: aborting due to previous error
```

alex, 2019-09-02:
What kernel version are you on?

geofft, 2019-09-02:
Ugh I think setfl might be a Debian-specific patch?? I can't find any mention of it in the upstream kernel, on my most convenient Debian box I see

```
#ifndef __GENKSYMS__
        /* bwh: For aufs only */
        int (*setfl)(struct file *, unsigned long);
#endif
```

and bwh is the Debian kernel maintainer, and aufs is out-of-tree.

geofft, 2019-09-02:
Thanks for the report @aurabindo - we'll figure out a way to conditionally compile this.

aurabindo, 2019-09-02:
I'm on `4.19.32`, archlinux

aurabindo, 2019-09-02:
Would a lame check like https://paste.ubuntu.com/p/tFC7JsXxjt/ do the job ?

alex, 2019-09-02:
We need to know about the kernel we're compiling for, not the one we're
running on, so no this isn't sufficient.

alex, 2019-09-12:
Is there a define or a config or anything like that for this?

I'm hitting this trying to do development on #139, and I'm very mad at computer :-)

closed 2019-09-15 by geofft

geofft, 2019-09-15:
@aurabindo, can you see if the latest commit on master works?

(Note that in the meantime we've switched from `cargo xbuild` to `cargo build` - take a look at the current README, and note you might need to `rustup update nightly` or something.)

aurabindo, 2019-09-16:
@geofft My nightly update is failing, and I do not have -Z flag `build-std`. Could you tell me your specific nightly version ?

geofft, 2019-09-16:
That flag was added a couple days ago, sorry :) Try `rustup update nightly`

aurabindo, 2019-09-16:
`rustup update nightly` is failing for me. I explicitly asked cargo compile with latest nightly. Even then its failing. 
```
diff --git a/hello-world/Kbuild b/hello-world/Kbuild
index d2114c4..78013af 100644
--- a/hello-world/Kbuild
+++ b/hello-world/Kbuild
@@ -4,7 +4,7 @@ helloworld-objs := hello_world.rust.o
 CARGO ?= cargo

 $(src)/target/x86_64-linux-kernel/debug/libhello_world.a: $(src)/Cargo.toml $(wildcard $(src)/src/*.rs)
-       cd $(src); env -u MAKE -u MAKEFLAGS $(CARGO) build -Z build-std=core,alloc --target=x86_64-linux-kernel
+       cd $(src); env -u MAKE -u MAKEFLAGS $(CARGO) +nightly-2019-09-16 build -Z build-std=core,alloc --target=x86_64-linux-kernel

 %.rust.o: target/x86_64-linux-kernel/debug/lib%.a
        $(LD) -r -o $@ --whole-archive $<
```

alex, 2019-09-16:
2019-09-11 nightly should work

commit f6db9710 (2019-09-15): Only use file_operations::setfl on AUFS-patched kernels (closes #167) (#182)

Debian and Ubuntu, which we're using for all our CI and local dev, have
a patched kernel which adds a setfl method to struct file_operations, so
our builds don't work on unpatched (e.g., upstream) kernels. The
patchset also does an EXPORT_SYMBOL_GPL(setfl), so we can key on that to
determine whether we need to fill in the setfl method or not.

Current patches are at
https://salsa.debian.org/kernel-team/linux/tree/debian/5.2.9-2/debian/patches/features/all/aufs5