fishinabarrel/linux-kernel-module-rust #173: Build fails on v5.3

kees, 2019-09-06:
I think the Kbuild targets changed. If I force ld to include the .a file, I can get further...

```
$ KDIR=~/src/linux make
make -C /home/kees/src/linux M=/home/kees/src/rust-linux/hello-world
make[1]: Entering directory '/home/kees/src/linux'
  LD [M]  /home/kees/src/rust-linux/hello-world/helloworld.o
ld: no input files
scripts/Makefile.build:471: recipe for target '/home/kees/src/rust-linux/hello-world/helloworld.o' failed
make[2]: *** [/home/kees/src/rust-linux/hello-world/helloworld.o] Error 1
Makefile:1624: recipe for target '_module_/home/kees/src/rust-linux/hello-world' failed
make[1]: *** [_module_/home/kees/src/rust-linux/hello-world] Error 2
make[1]: Leaving directory '/home/kees/src/linux'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2

```

alex, 2019-09-06:
I think this has roughly the same root cause as what I'm writing about here: https://github.com/fishinabarrel/linux-kernel-module-rust/pull/142#issuecomment-524297219

Trying to pass `.a` files where the kernel is expecting `.o` files is not a happy times, we need to make our build chain smarter and convert the `.a` file into a single `.o` file, which the kernel will like better.

geofft, 2019-09-07:
Wild guess, does the patch in #175 fix your problem?

kees, 2019-09-07:
It does! Thank you :)

closed 2019-09-07 by kees