fishinabarrel/linux-kernel-module-rust #174: Build fails with: ERROR: "mcount" [/home/kees/src/rust-linux/hello-world/helloworld.ko] undefined!

kees, 2019-09-06:
This is likely some kind of mismatch between gcc and clang when dealing with ftrace. Disabling CONFIG_FTRACE didn't help, though. :(

```
$ KDIR=~/src/linux make
make -C /home/kees/src/linux M=/home/kees/src/rust-linux/hello-world
make[1]: Entering directory '/home/kees/src/linux'
  LD [M]  /home/kees/src/rust-linux/hello-world/helloworld.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: modpost: missing MODULE_LICENSE() in /home/kees/src/rust-linux/hello-world/helloworld.o
see include/linux/module.h for more information
ERROR: "mcount" [/home/kees/src/rust-linux/hello-world/helloworld.ko] undefined!
scripts/Makefile.modpost:103: recipe for target 'modules-modpost' failed
make[2]: *** [modules-modpost] Error 1
Makefile:1627: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/home/kees/src/linux'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2

```

geofft, 2019-09-06:
Hmmm, my regular test box (stock Debian Buster) has CONFIG_FUNCTION_TRACER and CONFIG_FUNCTION_GRAPH_TRACER. But `nm helloworld.ko` indicates it's emitting a call to `__fentry__` and not to `mcount`, and /proc/kallsysms says my kernel is exporting a `__fentry__` too. My kernel has CONFIG_HAVE_FENTRY?

What kernel version are you on, what are the values of those three Kconfig options, what compiler was it built with, and what clang version are you using?

kees, 2019-09-07:
Building latest Linus kernel (v5.3-rc7), on Ubuntu Bionic.

```
$ grep CONFIG_FUNCTION_.*TRACER .config
$ grep CONFIG_HAVE_FENTRY .config
CONFIG_HAVE_FENTRY=y
$ gcc --version
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
...
$ clang --version
clang version 10.0.0 (https://github.com/llvm/llvm-project.git 84cb4b2b533c9b662707ad3f6500b39370923e0d)
...
```

kees, 2019-09-08:
Hah. Have the .a -> .o build fix, now my mcount yelling happens at module load time. ;)

```
[   19.890310] helloworld: loading out-of-tree module taints kernel.
[   19.891431] helloworld: module license 'unspecified' taints kernel.
[   19.892559] Disabling lock debugging due to kernel taint
[   19.897924] helloworld: Unknown symbol mcount (err -2)
```

I'll keep digging.

geofft, 2019-09-08:
Just as a baseline of comparison, helloworld works on stock Ubuntu Bionic and emits `__fentry__` only:
```
$ uname -a
Linux ip-172-16-0-188 4.15.0-1044-aws #46-Ubuntu SMP Thu Jul 4 13:38:28 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
$ grep CONFIG_FUNCTION.*TRACER /boot/config-"$(uname -r)"
CONFIG_FUNCTION_TRACER=y
CONFIG_FUNCTION_GRAPH_TRACER=y
$ grep CONFIG_HAVE_FENTRY /boot/config-"$(uname -r)"
CONFIG_HAVE_FENTRY=y
$ gcc --version
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
[...]
$ clang --version
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
[...]
$ nm helloworld.ko | egrep '(fentry|mcount)'
                 U __fentry__
$ egrep '(fentry|mcount)' /proc/kallsyms 
0000000000000000 T __fentry__
0000000000000000 R __ksymtab___fentry__
0000000000000000 r __kstrtab___fentry__
0000000000000000 T __start_mcount_loc
0000000000000000 T __stop_mcount_loc
```

Let me reboot into 5.3-rc7.

geofft, 2019-09-08:
Whoops apparently the upstream kernel doesn't have either disk or network drivers for AWS Nitro instances, I will have to recover my instance some other day :)

Anyway I guess my immediate questions are, does it work with an older clang (e.g. clang 9 from apt.llvm.org), and does it work with your clang and the stock kernel?

luisgerhorst, 2019-09-17:
This occured to me on Debian and I was just able to reproduce it on a freshly installed [Desktop Ubuntu 19.04](https://ubuntu.com/download/desktop). I only installed `clang-9` (as well as the packages suggested in the README).

luisgerhorst, 2019-09-18:
I managed to fix it by installing Ubuntu 18.04 Bionic which for me had a 5.0 kernel initially. With that kernel, it did not work (even after installing clang 9 from apt.llvm.org). Instead I had to [install](http://karlcode.owtelse.com/blog/2017/03/13/reverting-to-a-previous-kernel/) the `4.15.0-64-generic` kernel:

```sh
sudo apt install linux-image-4.15.0-64-generic
sudo apt install linux-headers-4.15.0-64-generic
```

I booted it (see [askubuntu.com](https://askubuntu.com/questions/314754/how-to-get-the-grub-using-virtualbox) for a guide on doing that in VirtualBox) and then did

```sh
cd hello-world
make clean # required!
make
```

p2mate, 2019-10-12:
Same problem happens on arch linux with kernel 5.3.5-arch1-1-ARCH. I solved it by modifying build.rs to filter out -pg when building src/helpers.c. Apparently the -pg argument in clang causes functions to call mcount() for profiling purposes. However gcc -mfentry to make it call __fentry__ earlier in the function. Arch linux seems to build their x86_64 kernels with this option. It seems clang does not support this however, causing problems when building helper.c with clang.

H-Plus-Time, 2019-10-13:
Another solution on Arch linux is to add '-mfentry' to the builder flags just before compiling 'src/helpers.c', completely removes the mcount symbol.

alex, 2019-12-30:
Ok, I'm seeing this as well. I think we need a real solution for this.

@geofft do you understand the underlying issue here?

0kk4cce1er, 2020-05-23:
I spent a lot of time, and now successfully run on ubuntu18.04, here are my changes.

https://gist.github.com/0kk4cce1er/b7a504fd676c1ef42de121b3ac2beeb3

alex, 2020-08-09:
@0kk4cce1er are you interested in sending a pull request for your changes?

geofft, 2020-08-09:
All right, I finally figured out what's going on.

This is all due to the fact that we're building modules using Clang (really bindgen, which uses libclang, but it's functionally equivalent to writing a C-language module and compiling it with Clang) and most people's kernels are built with GCC. The kernel has two ways of testing whether the current compiler is GCC or Clang. One happens at configuration time and is stored in the `CONFIG_CC_IS_GCC`/`CONFIG_CC_IS_CLANG` variables. One happens inside the Makefile by grepping the output of `${CC} --version` at build time.

kernel-cflags-finder sets `CC=clang`, so it causes the second mechanism to notice that we're Clang but not the first. So the kernel still generates some options that aren't valid for Clang.

The kernel's Makefile [uses feature detection to figure out if `-mfentry` exists in the compiler](https://github.com/torvalds/linux/blob/v5.8/Makefile#L852-L857). Because the kernel was generating options that aren't valid for Clang, feature detection was failing - any attempt to compile anything would get you an error about the GCC-only flags. So, the Makefile was wrongly concluding that the compiler didn't support `-mfentry`.

Thanks to @H-Plus-Time and @0kk4cce1er for pointing out that manually adding `-mfentry` _does_ work in Clang and does fix the problem, which was enough of a hint to start tracking down why the kernel build system didn't think it did.

#227 fixes this by setting `CONFIG_CC_IS_CLANG=y` in kernel-cflags-finder. That fixes the issue on a test Ubuntu Bionic VM and causes kernel-cflags-finder to output `-mfentry`.

There's a separate issue that kernel-cflags-finder should have been exiting with this same error (`ERROR: "mcount"  undefined!`) when building its dummy kernel module, but it wasn't quite doing so, probably because the "module" has no functions. I'll send in a followup fix to give the dummy module an empty `init_module` + `cleanup_module` + `MODULE_LICENSE("GPL")`, which should help us catch errors like this earlier. (That is, if it had failed there, it would have been more obvious that the issue is entirely with building Clang modules on a GCC kernel and didn't involve Rust at all.)

commit 0de435d2 (2020-08-09): kernel-cflags-finder: Set CONFIG_CC_IS_CLANG=y

Some of the things in the kernel's Makefile check whether
`${CC} --version` is clang, which is affected by setting CC. Some look at
CONFIG_CC_IS_CLANG, which was set at configuration time when our kernel
was built. We need to override the latter to get clang-compatible flags.

This causes the kernel to no longer set -W / -Wno flags that aren't
valid for clang, so we can get rid of `-Wno-unknown-warning-option`.

One problem with the previous approach is that the kernel's Makefile
does feature detection to see what options are accepted by the curr [... cut]
PR #227 (2020-08-09): kernel-cflags-finder: Set CONFIG_CC_IS_CLANG=y
Some of the things in the kernel's Makefile check whether
`${CC} --version` is clang, which is affected by setting CC. Some look at
CONFIG_CC_IS_CLANG, which was set at configuration time when our kernel
was built. We need to override the latter to get clang-compatible flags.

This causes the kernel to no longer set -W / -Wno flags that aren't
valid for clang, so we can get rid of `-Wno-unknown-warning-option`.

One problem with the previous approach is that the kernel's Makefile
does feature detection to see what options are accepted by the curre [... cut]
commit 490173c2 (2020-08-09): kernel-cflags-finder: Set CONFIG_CC_IS_CLANG=y (#227)

Some of the things in the kernel's Makefile check whether
`${CC} --version` is clang, which is affected by setting CC. Some look at
CONFIG_CC_IS_CLANG, which was set at configuration time when our kernel
was built. We need to override the latter to get clang-compatible flags.

This causes the kernel to no longer set -W / -Wno flags that aren't
valid for clang, so we can get rid of `-Wno-unknown-warning-option`.

One problem with the previous approach is that the kernel's Makefile
does feature detection to see what options are accepted by t [... cut]
closed 2020-08-09 by alex