fishinabarrel/linux-kernel-module-rust #219: hello-world build and run on Debian 5.6.14 kernel 5.6.0-2-amd64

BartMassey, 2020-06-13:
[This branch](https://github.com/BartMassey/linux-kernel-module-rust/tree/debian-5.6.14-1-linux-5.6.0-2-amd64) in my repo fork makes the `hello-world` example build and work on on Debian-5.6.14-1 with Linux kernel 5.6.0-2-amd64. I did not put in a pull request, because this was sort of kludged together. Thought someone might want it, though. I built using `clang-11`: `clang-9` did not work for me, but I don't know whether `clang-10` would have.

alex, 2020-07-12:
@BartMassey sorry I was so slow to look at this -- I think all of the type issues you had to fix should be resolved on `master` now -- can you take a look and see what's still required, and do a PR for that?

geofft, 2020-08-09:
#227 adds `CONFIG_CC_IS_CLANG`.

So it looks like the remaining diffs are
* `eh_personality` - this surprises me, we disable unwinding. How'd you run into this? (You might need a newer Rust nightly?)
* type issues - try rebasing?

BartMassey, 2020-08-10:
Current master  `86a8004` seems to compile and run fine on my box now. Requires `export CLANG=clang-11` — does not work with `clang-9` or `clang-10`. Suggest updating the README to suggest `clang-11` for latest 5.0 kernels.

Thanks much for this project, and for adding the requisite stuff to get it working again.

```
Linux redacted 5.7.0-2-amd64 #1 SMP Debian 5.7.10-1 (2020-07-26) x86_64 GNU/Linux
```

geofft, 2020-08-10:
What's your build failure on lower clang versions?

(I can _probably_ reproduce this myself, but I'm currently tracking down an issue where Ubuntu Bionic + clang 9 + kernel 4.15 fails on Travis but not on EC2 and I have yet to figure out what the relevant difference is, so it seems safer to ask...)

geofft, 2020-08-12:
Are you getting something like
```
arch/x86/include/asm/segment.h:266:2: error: expected '(' after 'asm'
```
? If so, that's probably https://reviews.llvm.org/D75563 yeah.

BartMassey, 2020-08-12:
Yes, that's the one.

geofft, 2020-08-13:
OK, I updated the README to mention it. Though.. we could work around it by adding a `-UCONFIG_CC_HAS_ASM_INLINE` in build.rs, which might be worthwhile since this version of clang is literally not released yet. (And hm I think it's actually a matter of what gcc version your kernel was built with.)

BartMassey, 2020-08-14:
I guess it depends on how close we are to the clang-11 release, which I'm not really tracking. I suspect pretty close, since Debian is packaging it in unstable at this point.