fishinabarrel/linux-kernel-module-rust #223: Golang support

ski7777, 2020-07-13:
I hope this isn't too off-topic. If yes, feel free to delete it.
Do you think it would be possible to do the same in golang? If yes, what would be the minimal requirements to create a simple hello world? I don't understand much of rust programming but as far as I know go and rust seem to have a quite similar architecture so I hope it should be possible to write a kernel module in golang.

Regards
Raphael

alex, 2020-07-13:
This is well beyond the scope for this project.

From my perspective, Go is likely to be more complex than Rust for kernel interop, since it has a much more significant runtime (including garbage collector). I have no idea if there's any prior art here.

closed 2020-07-13 by alex

geofft, 2020-07-14:
Right, the Go runtime also manages goroutines, which are lightweight threads that generally assume something will handle OS-level threading for-it. In the kernel nothing will handle threading for you. :)

Go also doesn't have as direct interoperability with C - cgo calls are high-overhead as Go adapts how it uses the stack and other things to the way that C expects. Rust's use of the stack etc. (its calling convention) is pretty close to the native C calling convention - the difference is mostly in passing Rust types that don't exist in C - so you can easily have C code that calls into Rust that calls into C, and we rely on that. It's going to be difficult to write a Go kernel module that takes advantage of existing kernel functionality and also provides the levels of performance people expect out of a compiled language.

So, if you try, you could probably get _some_ Go code running in the kernel, but it wouldn't have the full power of normal Go and it wouldn't be something you'd want to use for real-world use cases, probably.