fishinabarrel/linux-kernel-module-rust #225: Error with kernel-cflags-finder on make

kilroyjones, 2020-07-20:
I'm incredibly new to all this, so pardon my ignorance, but I was trying to compile the hello-world example and experience this error using Mint 20 with kernels 5.4.0.26, 5.6.19, and 5.7.9.

```
   Compiling bitflags v1.2.1
   Compiling linux-kernel-module v0.1.0 (/home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust)
error: failed to run custom build command for `linux-kernel-module v0.1.0 (/home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust)`

Caused by:
  process didn't exit successfully: `/home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/hello-world/target/debug/build/linux-kernel-module-072fd5c94419c165/build-script-build` (exit code: 1)
  --- stdout   Compiling bitflags v1.2.1
   Compiling linux-kernel-module v0.1.0 (/home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust)
error: failed to run custom build command for `linux-kernel-module v0.1.0 (/home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust)`

Caused by:
  process didn't exit successfully: `/home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/hello-world/target/debug/build/linux-kernel-module-072fd5c94419c165/build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=KDIR
  cargo:rerun-if-env-changed=CLANG
  cargo:rerun-if-changed=kernel-cflags-finder/Makefile

  --- stderr
  kernel-cflags-finder did not succeed
  stdout: 
  stderr: error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'? [-Werror,-Wunknown-warning-option]
  make[2]: *** [scripts/Makefile.build:275: /home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/kernel-cflags-finder/dummy.o] Error 1
  make[1]: *** [Makefile:1719: /home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/kernel-cflags-finder] Error 2
  make: *** [Makefile:33: all] Error 2

make[2]: *** [/home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/hello-world/Kbuild:7: /home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/hello-world/target/x86_64-linux-kernel/debug/libhello_world.a] Error 101
make[1]: *** [Makefile:1719: /home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/hello-world] Error 2
make: *** [Makefile:4: all] Error 2
  cargo:rerun-if-env-changed=KDIR
  cargo:rerun-if-env-changed=CLANG
  cargo:rerun-if-changed=kernel-cflags-finder/Makefile

  --- stderr
  kernel-cflags-finder did not succeed
  stdout: 
  stderr: error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'? [-Werror,-Wunknown-warning-option]
  make[2]: *** [scripts/Makefile.build:275: /home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/kernel-cflags-finder/dummy.o] Error 1
  make[1]: *** [Makefile:1719: /home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/kernel-cflags-finder] Error 2
  make: *** [Makefile:33: all] Error 2

make[2]: *** [/home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/hello-world/Kbuild:7: /home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/hello-world/target/x86_64-linux-kernel/debug/libhello_world.a] Error 101
make[1]: *** [Makefile:1719: /home/kilroy/code/kernel_modules/rust_based/linux-kernel-module-rust/hello-world] Error 2
make: *** [Makefile:4: all] Error 2
```
I also tried using this [branch](https://github.com/BartMassey/linux-kernel-module-rust/tree/debian-5.6.14-1-linux-5.6.0-2-amd64) and have followed all the recommend changes both [here](https://github.com/lizhuohua/linux-kernel-module-rust/issues/1) and [here](https://github.com/lizhuohua/linux-kernel-module-rust/issues/1). It seemed that some of these have already been changed here, such as cflags. 

Any advice would be much appreciated.

alex, 2020-07-20:
Interesting. What clang version were you using?

kilroyjones, 2020-07-20:
Tried it with clang 9 and 10, and both produced the same error.

alex, 2020-07-21:
Mmm, not sure off hand. @geofft any idea?

geofft, 2020-08-09:
Oh, this is probably also fixed by #227!

alex, 2020-08-09:
Optimistically closing this now that #227 is merged. If it's not actually fixed, we can reopen!

closed 2020-08-09 by alex