fishinabarrel/linux-kernel-module-rust #242: Build fails on Debian stable, kernel 4.19: implicit declaration of function 'asm_volatile_goto'

geofft, 2020-08-13:
On Debian 10's kernel 4.19, with any of clang 7 (Debian stable), 9 (apt.llvm.org prerelease, 20190716095603) or 12 (apt.llvm.org prerelease, 20200812102626):

```
root@lkm:~/linux-kernel-module-rust/hello-world# make CLANG=clang-12
make -C /lib/modules/4.19.0-5-amd64/build M=/root/linux-kernel-module-rust/hello-world CC=clang-12 CONFIG_CC_IS_CLANG=y
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-5-amd64'
cd /root/linux-kernel-module-rust/hello-world; cargo build -Z build-std=core,alloc --target=x86_64-linux-kernel
   Compiling linux-kernel-module v0.1.0 (/root/linux-kernel-module-rust)
error: failed to run custom build command for `linux-kernel-module v0.1.0 (/root/linux-kernel-module-rust)`

Caused by:
  process didn't exit successfully: `/root/linux-kernel-module-rust/hello-world/target/debug/build/linux-kernel-module-b85ad71b343c44f6/build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=CC
  cargo:rerun-if-env-changed=abs_srctree
  cargo:rerun-if-env-changed=c_flags
  cargo:rerun-if-changed=src/bindings_helper.h

  --- stderr
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:36:2: error: implicit declaration of function 'asm_volatile_goto' [-Wimplicit-function-declaration]
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:42:3: error: expected ')'
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:36:19: note: to match this '('
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:51:2: error: implicit declaration of function 'asm_volatile_goto' [-Wimplicit-function-declaration]
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:58:3: error: expected ')'
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:51:19: note: to match this '('
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/cpufeature.h:164:2: error: implicit declaration of function 'asm_volatile_goto' [-Wimplicit-function-declaration]
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/cpufeature.h:195:4: error: expected ')'
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/cpufeature.h:164:19: note: to match this '('
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:36:2: error: implicit declaration of function 'asm_volatile_goto' [-Wimplicit-function-declaration], err: true
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:42:3: error: expected ')', err: true
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:51:2: error: implicit declaration of function 'asm_volatile_goto' [-Wimplicit-function-declaration], err: true
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/jump_label.h:58:3: error: expected ')', err: true
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/cpufeature.h:164:2: error: implicit declaration of function 'asm_volatile_goto' [-Wimplicit-function-declaration], err: true
  /usr/src/linux-headers-4.19.0-5-common/arch/x86/include/asm/cpufeature.h:195:4: error: expected ')', err: true
  thread 'main' panicked at 'Unable to generate bindings: ()', /root/linux-kernel-module-rust/build.rs:177:39
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
make[4]: *** [/root/linux-kernel-module-rust/hello-world/Kbuild:10: /root/linux-kernel-module-rust/hello-world/target/x86_64-linux-kernel/debug/libhello_world.a] Error 101
make[3]: *** [/usr/src/linux-headers-4.19.0-5-common/Makefile:1539: _module_/root/linux-kernel-module-rust/hello-world] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-5-amd64'
make: *** [Makefile:9: all] Error 2
```

geofft, 2020-08-13:
That's because 4.19 is missing torvalds/linux@8bd66d147c88bd441178c7b4c774ae5a185f19b8 (fixed inv4.20). Anything other than 4.19.x should be fine.

It's a small fix, maybe we should just inline it?

geofft, 2020-08-13:
Bleh, the kernel build system generates a .mod.c file that's also impacted and patching that is less obvious.

alex, 2020-08-13:
@geofft is asking a debian friend to backport this to their kernel a crazy idea?

geofft, 2020-08-14:
Maybe it should really be in upstream 4.19 stable? @nickdesaulniers / @ojeda, is it reasonable to get torvalds/linux@8bd66d147c88bd441178c7b4c774ae5a185f19b8 in -stable?

ojeda, 2020-08-14:
It is already there since v4.19.48, see https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=fd45cd4530ebc7c846f83b26fef526f4c960d1ee

geofft, 2020-08-14:
Ah, yes, it worked after updating my VM. Debian 4.19.0-5 is 4.19.37 and 4.19.0-10 is 4.19.132. Thanks 🤦

closed 2020-08-14 by geofft

ojeda, 2020-08-14:
You're welcome!

By the way, if this is for Plumbers, then do not worry much about older kernels, showing it under e.g. v5.8 is more than good enough.

geofft, 2020-08-14:
Nah, Plumbers is just motivation to try to keep this repo in good shape, and the changes I landed earlier this week ended up breaking this test VM. Mostly I want 4.19 working because it's the default for Debian stable and I want things to work for people who come by and check out this repo and are using their distro kernel, and a bit I want this working because we use 4.19 at work :)

I do want to set up CI for Linux HEAD though ... I've been looking at how clangbuiltlinux/continuous-integration does things. (I don't suppose it saves its built kernels anywhere, does it?)

ojeda, 2020-08-14:
I think @nathanchance is the one with most experience on the CI over there.

nathanchance, 2020-08-18:
Sorry for taking a bit to get back to this.

> I do want to set up CI for Linux HEAD though ... I've been looking at how clangbuiltlinux/continuous-integration does things. (I don't suppose it saves its built kernels anywhere, does it?)

We do not save any kernels since we do not use CI to distribute anything. Not sure if that would be the goal of your own CI but we just want to run some simple build tests to make sure that upstream LLVM and Linux do not break anything (and of course, everything is broken right now so we are red all over the place...).

I am not sure how you would go about hooking in an upload step but I am sure it cannot be too difficult since we do that for our Docker image.

geofft, 2020-08-18:
Mostly I want to run CI on top of kernels from HEAD and want to avoid compiling them if I can :) but I can just fork your travis.yml and have it build kernels and then build our modules on top, that would certainly work too.