fishinabarrel/linux-kernel-module-rust #255: rust nightly clang-12 on debian unstable 5.7.0-3-amd64 builds and runs successfully

omac777, 2020-08-27:
I wanted to drop by and say it builds and runs.  I also included a few notes to setup in case others want to repeat the successful build and run.

```
uname -r
5.7.0-3-amd64

uname -a
Linux righteous 5.7.0-3-amd64 #1 SMP Debian 5.7.17-1 (2020-08-23) x86_64 GNU/Linux

apt-get install linux-headers-"$(uname -r)"
apt-get install clang-12 llvm-12 lldb-12 clang-tools-12 llvm-12-tools clang-12-examples clang-12-doc libomp-12-doc llvm-12-doc
update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-12/bin/clang++ 1001
update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-12/bin/clang 1001
sudo update-alternatives --config clang
sudo update-alternatives --config clang++
```

if things don't look right tweak debian alternatives files directly
```
cd /var/lib/dpkg/alternatives/ 
emacs -nw clang
emacs -nw clang++
```

```
cat /var/lib/dpkg/alternatives/clang
auto
/usr/bin/clang

/usr/lib/llvm-10/bin/clang
1000
/usr/lib/llvm-12/bin/clang
1001
/usr/lib/llvm-7/bin/clang
100
```

```
cat /var/lib/dpkg/alternatives/clang++
auto
/usr/bin/clang++

/usr/lib/llvm-10/bin/clang++
1000
/usr/lib/llvm-12/bin/clang++
1001
/usr/lib/llvm-7/bin/clang++
100
```

then check once again everything looks good
```
sudo update-alternatives --config clang++
sudo update-alternatives --config clang++
```

Here's the build and run output in detail for those interested:
```
loongson@righteous 2020-08-27_07:28:07_EDT : ~/redoxstuff
$ git clone https://github.com/fishinabarrel/linux-kernel-module-rust.git
Cloning into 'linux-kernel-module-rust'...
remote: Enumerating objects: 44, done.        
remote: Counting objects: 100% (44/44), done.        
remote: Compressing objects: 100% (36/36), done.        
remote: Total 1721 (delta 11), reused 21 (delta 2), pack-reused 1677        
Receiving objects: 100% (1721/1721), 252.09 KiB | 6.81 MiB/s, done.
Resolving deltas: 100% (1012/1012), done.
loongson@righteous 2020-08-27_07:28:42_EDT : ~/redoxstuff
$ cd linux-kernel-module-rust/hello-world/
loongson@righteous 2020-08-27_07:28:46_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world
$ rustup component add --toolchain=nightly rust-src rustfmt
info: component 'rust-src' is up to date
info: component 'rustfmt' for target 'x86_64-unknown-linux-gnu' is up to date
loongson@righteous 2020-08-27_07:29:11_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world
$ clang --version
Debian clang version 12.0.0-++20200826105036+d7c119d89c5-1~exp1~20200826205657.3447
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
loongson@righteous 2020-08-27_07:29:39_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world
$ clang++ --version
Debian clang version 12.0.0-++20200826105036+d7c119d89c5-1~exp1~20200826205657.3447
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
loongson@righteous 2020-08-27_07:29:45_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world
$ rustc --version
rustc 1.45.0-nightly (5fd2f06e9 2020-05-31)
loongson@righteous 2020-08-27_07:29:54_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world
$ make
make -C /lib/modules/5.7.0-3-amd64/build M=/home/loongson/redoxstuff/linux-kernel-module-rust/hello-world CC=clang CONFIG_CC_IS_CLANG=y
make[1]: Entering directory '/usr/src/linux-headers-5.7.0-3-amd64'
  AR      /home/loongson/redoxstuff/linux-kernel-module-rust/hello-world/built-in.a
cd /home/loongson/redoxstuff/linux-kernel-module-rust/hello-world; cargo build -Z build-std=core,alloc --target=x86_64-linux-kernel
    Updating crates.io index
 Downloading crates ...
  Downloaded lazycell v1.3.0
  Downloaded bindgen v0.54.0
   Compiling memchr v2.3.3
   Compiling libc v0.2.76
   Compiling bitflags v1.2.1
   Compiling version_check v0.9.2
   Compiling cc v1.0.59
   Compiling glob v0.3.0
   Compiling log v0.4.11
   Compiling compiler_builtins v0.1.31
   Compiling proc-macro2 v1.0.19
   Compiling lazy_static v1.4.0
   Compiling unicode-xid v0.2.1
   Compiling regex-syntax v0.6.18
   Compiling unicode-width v0.1.8
   Compiling cfg-if v0.1.10
   Compiling quick-error v1.2.3
   Compiling strsim v0.8.0
   Compiling vec_map v0.8.2
   Compiling termcolor v1.1.0
   Compiling bindgen v0.54.0
   Compiling ansi_term v0.11.0
   Compiling core v0.0.0 (/home/loongson/.rustup/toolchains/nightly-2020-06-01-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/src/libcore)
   Compiling rustc-hash v1.1.0
   Compiling peeking_take_while v0.1.2
   Compiling lazycell v1.3.0
   Compiling shlex v0.1.1
   Compiling thread_local v1.0.1
   Compiling textwrap v0.11.0
   Compiling humantime v1.3.0
   Compiling nom v5.1.2
   Compiling clang-sys v0.29.3
   Compiling aho-corasick v0.7.13
   Compiling libloading v0.5.2
   Compiling quote v1.0.7
   Compiling atty v0.2.14
   Compiling which v3.1.1
   Compiling clap v2.33.3
   Compiling regex v1.3.9
   Compiling cexpr v0.4.0
   Compiling env_logger v0.7.1
   Compiling linux-kernel-module v0.1.0 (/home/loongson/redoxstuff/linux-kernel-module-rust)
   Compiling rustc-std-workspace-core v1.99.0 (/home/loongson/.rustup/toolchains/nightly-2020-06-01-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/src/tools/rustc-std-workspace-core)
   Compiling alloc v0.0.0 (/home/loongson/.rustup/toolchains/nightly-2020-06-01-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/src/liballoc)
   Compiling hello-world v0.1.0 (/home/loongson/redoxstuff/linux-kernel-module-rust/hello-world)
    Finished dev [unoptimized + debuginfo] target(s) in 33.52s
ld -r -o /home/loongson/redoxstuff/linux-kernel-module-rust/hello-world/hello_world.rust.o --whole-archive /home/loongson/redoxstuff/linux-kernel-module-rust/hello-world/target/x86_64-linux-kernel/debug/libhello_world.a
  LD [M]  /home/loongson/redoxstuff/linux-kernel-module-rust/hello-world/helloworld.o
  MODPOST 1 modules
  CC [M]  /home/loongson/redoxstuff/linux-kernel-module-rust/hello-world/helloworld.mod.o
  LD [M]  /home/loongson/redoxstuff/linux-kernel-module-rust/hello-world/helloworld.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.7.0-3-amd64'
loongson@righteous 2020-08-27_07:30:55_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world

$ sudo insmod helloworld.ko
[sudo] password for loongson: 
loongson@righteous 2020-08-27_07:31:34_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world

$ sudo rmmod helloworld
loongson@righteous 2020-08-27_07:32:06_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world

$ sudo dmesg | tail
[ 4010.241390] helloworld: loading out-of-tree module taints kernel.
[ 4010.243641] helloworld: module verification failed: signature and/or required key missing - tainting kernel
[ 4010.254851] Hello kernel module!
[ 4042.212931] My message is on the heap!
[ 4042.212933] Goodbye kernel module!
loongson@righteous 2020-08-27_07:32:27_EDT : ~/redoxstuff/linux-kernel-module-rust/hello-world
```

geofft, 2020-08-27:
Neat! You should also be able to set the CLANG environment variable instead of doing update-alternatives, e.g., `CLANG=clang-12 make`.

alex, 2020-08-29:
Awesome! I'm going to close this though, since there's no action to be taken.

closed 2020-08-29 by alex