fishinabarrel/linux-kernel-module-rust #260: Add a nifty virtio driver

geofft, 2020-08-28:
@joshtriplett proposes that the first driver we try to get into the upstream kernel should be some neat virtio thing, alongside an implementation in qemu (and maybe also one of the virtio hypervisors in Rust!). This would have the benefit of letting people play with it in a VM and also make it non-critical, so we can enable it in defconfig if you have Rust nightly installed and are on a compatible arch (in particular, we're neither duplicating nor replacing an existing driver).

Maybe there's some interesting channel we can virtualize - the concept of virtio-wayland was mentioned as inspiration.

geofft, 2020-08-28:
A couple of options:
* virtio-midi - it's a pretty simple protocol, it'll be fun, it won't be in anyone's way
* Google has a virtio-based Android "Cuttlefish" platform for testing, running on crosvm, their maintainers might have more virtio things they'd like support for

stappersg, 2020-10-25:
_Dropping other options_

* `virtio-serial`   implements a serial port.
* `virtio-netrir`  network device re-implemented in Rust.

Thing to keep in mind is that the Rust kernel module in the QEMU guest also will need a matching counter part in the QEMU host.

From https://www.redhat.com/en/blog/virtio-devices-and-drivers-overview-headjack-and-phone do I understand that _PCI card emulation_ is in play.  That would mean that giving the QEMU guest a `virtio-netrir` network card wouldn't require building a QEMU host counterpart.

While writing down that, came this to mind:
* `modprobe virtio-netrir` will be like inserting a PCI  NIC.
* `rmmod virtio-nettrir` will be like removing a PCI Network Interface Card.