fishinabarrel/linux-kernel-module-rust #275: failed to build in Fedora32 due to expected '(' after 'asm'

YsuOS, 2020-10-05:
Hi

I failed to build `hello-world` in Fedora32(5.7.17-200.fc32.x86_64).

My clang version is `Fedora 10.0.0-2.fc32`.

```
# make
...
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s                                                                                                
  MODPOST 1 modules                                                           
  CC [M]  /root/share/linux-kernel-module-rust/hello-world/helloworld.mod.o                                                                                  
In file included from /root/share/linux-kernel-module-rust/hello-world/helloworld.mod.c:1:                                                                   
In file included from ./include/linux/module.h:13:                            
In file included from ./include/linux/stat.h:19:                              
In file included from ./include/linux/time.h:6:                                                                                                              
In file included from ./include/linux/seqlock.h:36:                           
In file included from ./include/linux/spinlock.h:51:                          
In file included from ./include/linux/preempt.h:78:                      
In file included from ./arch/x86/include/asm/preempt.h:7:                                                                                                    
In file included from ./include/linux/thread_info.h:38:                       
In file included from ./arch/x86/include/asm/thread_info.h:12:                                                                                               
In file included from ./arch/x86/include/asm/page.h:12:                                                                                                      
./arch/x86/include/asm/page_64.h:49:2: error: expected '(' after 'asm'                                                                                       
        alternative_call_2(clear_page_orig,                                                                                                                  
        ^                                                                                                                                                    
./arch/x86/include/asm/alternative.h:256:2: note: expanded from macro 'alternative_call_2'                                                                   
        asm_inline volatile (ALTERNATIVE_2("call %P[old]", "call %P[new1]", feature1,\
        ^                                                                                                                                                    
./include/linux/compiler_types.h:201:24: note: expanded from macro 'asm_inline'                                                                              
#define asm_inline asm __inline                                               
                       ^
...
```

As shown above error messages, Clang don't support and didn't expect "asm_inline" which is introduced from v5.4-rc1
so the CONFIG_CC_HAS_ASM_INLINE should be set "n".

If I replace asm_inline with asm in the kernel headers, it works fine.
I don't know how to set CONFIG_CC_HAS_ASM_INLINE when building the module.

So I just report this issue and let me ask your elegant workaround here.

reference

[x86: alternative.h: use asm_inline for all alternative variants · torvalds/linux@40576e5](https://github.com/torvalds/linux/commit/40576e5e63ea5eeab814de7af83ad09dd64134ee)

[linux/compiler_types.h at v5.4-rc2 · torvalds/linux](https://github.com/torvalds/linux/blob/v5.4-rc2/include/linux/compiler_types.h#L209-L213)

stappersg, 2020-11-13:
Interesting problem.

Do known that you are _going where no human has gone before_

Sorry for providing **only moral** support.

stappersg, 2020-11-14:
And today I found that [simulair error](https://github.com/fishinabarrel/linux-kernel-module-rust/issues/219#issuecomment-672649519) has been reported.