fishinabarrel/linux-kernel-module-rust #279: support extending bpf

dvc94ch, 2021-02-01:
Hi,
I watched your presentation from 2019 on youtube and you were asking about potential use cases for out of tree modules. I'd be interested in extending the bpf vm in out of tree kernel modules, but I'm not sure if that is possible. Specifically add support for unwinding user space stacks without frame pointers. Some related work I've done for creating better rust program analysis tools are in [0] and [1].

- [0] https://github.com/dvc94ch/rust-bpf
- [1] https://github.com/dvc94ch/ehframe-bpf-compiler

alex, 2021-02-01:
Hi David,

First: New development is happening at
https://github.com/Rust-for-Linux/linux/

Second: The first question to ask is whether ther kernel has existing APIs
for this. Thus far our focus has been on enabling folks to do things in
Rust, that can already be done in C. Discussions of _new_ kernel
functionality need to start upstream, not with us.

dvc94ch, 2021-02-01:
So this could fundamentally not be built using an out of tree kernel module? Sorry I don't know anything about kernel development.

alex, 2021-02-01:
I have no idea if the kernel supports extending BPF in the way you suggest.

dvc94ch, 2021-02-01:
Ok thanks

closed 2021-02-01 by dvc94ch